clough42/electronic-leadscrew #260: firmware please help

freddygino, 2023-05-10:
hi im trying to load the firmware on the launch pad followed the video to debug all went well till i hit the spider button i get load program error ...please help not to familiar to programming so this is all new to me

here is a pic of the error
[image]

tia

Chucksbp, 2023-05-13:
I have seen other people having a problem that totally removed code composer from their computer, then reinstalled code composer and re configured everything. It may be worth a try??  also i would go through the three switches and flick them back and fort a few times. the plastic tape they use can be very finicky and stop the switch movement. Again I am no expert and i found this to be the most troubling part of the build.  i can only say the end is well worth the trouble, you will love the ELS!

freddygino, 2023-05-13:
I tried about 6 different versions of ccs and all came back with the same error... I didn't try flicking the switches I'll try that... I did email support at Texas instruments and the have a section about that but I don't know what I'm reading cause I don't know coding lol

Chucksbp, 2023-05-13:
Are you using a version 9 of code composer? I know the latest versions will not work with Clough 42 software! Sent from my Galaxy

freddygino, 2023-05-13:
Yes I'm using the same version as James v9. 1

Chucksbp, 2023-05-13:
i had good luck with version 9.3.0

freddygino, 2023-05-13:
Tried that 1

freddygino, 2023-05-13:
this is what texas instruments sent me to resolve my issue...i really dont know what im reading ...if anyone would be so kind and checking out this link and help me with this i will greatly appreciate it ....thank you

https://e2e.ti.com/support/tools/code-composer-studio-group/ccs/f/code-composer-studio-forum/738865/faq-ccs-why-am-i-getting-a-data-verification-error-when-i-load-my-program

Chucksbp, 2023-05-14:
Can you post a picture of your Configuration H page? are there any errors showing in the box on the very bottom right of the page?

freddygino, 2023-05-14:
i installed ccs v9.3 and this is the error i get
[image]

freddygino, 2023-05-14:
this is the config h
[image]
[image]
[image]
[image]

Chucksbp, 2023-05-14:
Try this, remove the Clough42 board from the TI launch board. re install the three jumpers. use the supplied USB cable that came with the Launch board and see if you can program the chip. Thats how i did both of my boards. then i reinstalled the Clough42 board and connected everything. , 
i had issues with my first encoder and it wiped out encoder input one so i reprogramed the board several times successfully and then reprogramed a second board for a backup.

freddygino, 2023-05-14:
ok will try thanks

freddygino, 2023-05-14:
same error

Chucksbp, 2023-05-14:
i have no idea what is happening. Hopefully James will look at this and respond. he often does. I wish i could help more.

freddygino, 2023-05-14:
oh i so appreciate you try helping me thank you ,,,

Chucksbp, 2023-05-14:
you are welcome!!!

tomvikse, 2023-05-14:
Try a different computer..

freddygino, 2023-05-14:
Yea that's what I'm doing next thanks

freddygino, 2023-05-14:
tried different computer same thing....could there be something wrong with the file? i downloaded on 2nd computer from github
[image]

Chucksbp, 2023-05-15:
Are you trying to flash the chip from this view?? this is where you should be doing the flash from. , show me this view let me see if there are any errors in the bottom right. 
[image]

freddygino, 2023-05-15:
hi chuck yesterday i was looking at the board and noticed f280025c its the wrong board i checked my invoice and it says f280049c so i spoke with someone from mouser electronics the place i bought the board they are sending me the board tommorow...i apoligize hope its the right 1 tommorow

freddygino, 2023-05-15:
[image]

Chucksbp, 2023-05-15:
No need to apologies for anything!!!!  that's totally wild! who would have ever thought. At a glance it looks like the correct board. I hope this solves your problem, let us know if you get it working. best of luck to the finish!!!! I hope this fixes your issues!!!

freddygino, 2023-05-15:
thank you so much and definitly will let you know

kwackers, 2023-05-16:
I suspected that would be the case - hence my initial response... ;-)
Live and learn.

freddygino, 2023-05-16:
> I suspected that would be the case - hence my initial response... ;-) Live and learn.

Yes that was you first question I overlooked... Thank you

freddygino, 2023-05-16:
up and running in 3 minutes thanks guys

Chucksbp, 2023-05-17:
That's great!! Congratulations!!!!

freddygino, 2023-05-17:
> That's great!! Congratulations!!!!

thank you!!

freddygino, 2023-06-06:
hi guy finally got the els installed 2 problems if anyone can help....
1..on the control panel its on fwd but the lead screw is going oppisite

2...when i power off the control panel the lead screw is supposed to move free but it feels like the servo is still energized

tia

johnsonm, 2023-06-06:
You need to uncomment or comment (depending on what you have now)
```C
#define INVERT_DIRECTION_PIN true
```

What servo do you have? The iSVT7T, for example, doesn't have a enable line and so can't power on/off from the enable line, so doesn't have that feature.

ggwarrier, 2023-06-08:
@Chucksbp   I tried everywhich way to program the board without luck.  I even blew the booster board (3.3v went kaput) which took out the TI board.  Finally I tried your suggestion.  Removed the booster board, replaced the jumpers and 1.40 worked with CCS9.1.0
Thank you, Sir

glennto, 2023-10-14:
I have purchased a Launchpad TMS320F2800C, the Clough els booster board and Clough's remake of the LED/Key board (all worthwhile if you are considering replicating this project).  My goal was to get els 1.4 compiled and installed on the LaunchPad using the current/last Eclipse based TI CCSTUDIO IDE environment (version 12.4.0 or 12.5.0) - but not the new Theia IDE which looks too transitional to me.   This project under the version 9 IDE is aging out and I did not want to invest time in what is now becoming legacy software. As an example, newer versions of the IDE will automatically update the XDS 110 USB probe firmware making it incompatible with IDE V9 - I do not like dealing with this kind of stuff.  However to accomplish the update I had to learn the ins and outs of the whole TI IDE framework in parallel with dealing with els project compile and link errors.  In hindsight, the actual task was pretty straightforward - I just kept hammering away at each compile/link error as they came up.  For example, all of the supporting TI library functions are relocated in the newer versions installation directories and the build configuration include locations needed to be adjusted accordingly.  There seems to be an error in the argument typing for a TI furnished dma channel 6 initialization method and prototype definition that I fixed. TI has apparently dropped/renamed the all important #define _FLASH directive (to what I haven't figured out, but it works to just include this define in configure.h when you need it). There may have been other issues as well. I hope this post helps others.