clough42/electronic-leadscrew #261: Powering off the stepper motor while spindle is turning

ibonneau, 2023-05-13:
Hi,
Most of the time I don't want the stepper motor to turn. It's just noisy. It's annoying to turn off the lathe just to power down the stepper motor.

As I understood the idea behind locking the PWR button is electrical interference safety issue - which I have not encountered even in the first release.
Could summer assist me with enabling it out alternatively add a physical power button that would deactivate the stepper motor regardless if the spindle is turning or not without burning anything on the electronic boards or stepper motor.

🙏

kwackers, 2023-05-13:
If you turn off the controller using the power button on the far left it should release the enable line on the stepper driver and the motor shouldn't run.

ibonneau, 2023-05-13:
> If you turn off the controller using the power button on the far left it should release the enable line on the stepper driver and the motor shouldn't run.

That only works when the spindle is not rotating. 
I'd like the power button to turn it off regardless if the spindle is rotating or not - if that poses any saftey issue than a phycial switch to cut the enable lie would work without damaging the driver\stepper\boards ?

kwackers, 2023-05-13:
I guess we could ignore the fact you could have turned off the ELS before you started the spindle... ;) 

In UserInterface.cpp the power toggle occurs after the test for the spindle rotating.
Look in the function 'mainLoop' and you'll see an "if ( currentRpm == 0)" followed by "if ( keys.bit.POWER )"
If you move the test for the power button to before the test for currentRPM then you'll be able to turn it off even if the spindle is rotating.

I've obviously not tested this to see if there are any unintended consequences.

Failing that just fit a switch to the enable line.

ibonneau, 2023-05-13:
> power

Thank you.
Is there a chance that putting a physical switch on the enable line while everything else is running cause damage to the ti board the stepper or the driver?

I burnt a ti board a while back due to over voltage on the input by mistake and don't want to buy a third one🤣

kwackers, 2023-05-13:
Nope, all that will happen is the motor will stop dead.
Obviously make sure the enable line from the TI board can't get 'back-fed' with anything from your switch. If you're not keeping the enable from the TI board then it won't matter, otherwise I'd be inclined to use a break-before-make type switch.

(Be a lot easier just to tweak the software though).