clough42/electronic-leadscrew #262: Installation on Jet GH1340 - separate threading and feed screws

patl1, 2023-05-28:
I am ordering parts now to build the ELS for the Jet GH1340 lathe. The threading shaft is separate from the lead screw shaft and both are configured with several gear selection levers. II haven't found a way to set the various gear combinations on the lathe to have then turn at exactly the same ratio of input revs to output revs, so I would need to determine  a way (maybe my code modification) to set ratio driven parameters for the normal lead screw shaft, and a different ratio for the threading shaft.  I searched the forum and did not find anyone else describing this type of lathe configuration and wondered if anyone had solved this type of situation.

johnsonm, 2023-05-28:
You have to work it out either from the counts of all the gears (which might include opening up your gearbox) or experimentally (turning and measuring travel with a dial indicator) and factor it into your `STEPPER_RESOLUTION`

I tried using direct drive for my G0709, which also has separate lead screw and feed rod, and apparently I'm not alone in discovering that the feed rod is stiff to turn. It's so stiff that even if the feed is not engaged, my iSV57T-180 servo trips overcurrent if I select a gearbox setting that turns the feed rod, at least at all the gearbox settings I've tried so far.

Therefore, I'm currently working out modifying it to use a reduction outside the gearbox. The alternative is to remove the gearbox entirely and fabricate a bracket to hold the servo in place of the entire gearbox.

patl1, 2023-05-28:
I did measure the various ranges of movements on both shafts per rev. The threading shaft ranges from .035 to .250" per rev,  depending on the gear combinations. The feed shaft ranges from .0048 to .0271.  Unfortunately, no overlap in the ranges.  I am thinking of driving the shafts directly on the other end of the lathe and tying them together with the drive belts. Not my favorite solution because the servo motor and drive belts are all exposed and a little in the way, but I am not seeing a better solution yet. I was hoping that I could switch between ranges with button pushes on the LED pad (of course software/firmware mods would be needed). I'll keep cogitating it to see if I can dream up a more elegant solution.

johnsonm, 2023-05-29:
At least on the kwackers branch, you can have a separate feed rate with `STEPPER_RESOLUTION_FEED` — I don't remember if that's in the official branch though. You don't want to try to make the feed screw turn too fast.

Yes, I've also thought about putting a timing belt on the outboard end to tie the two together. That would make the clutch on the feed rod not work, but probably the belt would give way as long as I didn't overdo the belt. With `STEPPER_RESOLUTION_FEED` you can just make that 1:1.

If you haven't yet seen the kwackers branch, see PR #232 

Almost always for us hobbyists, feed can be close as long as threading is exactly right.

patl1, 2023-05-29:
I haven't dissected the code yet, however, I wonder if it is relatively easy to add another option to the button selection to reference a different ratio parameter variable in the code that is calibrated for the feed shaft rather than the threading shaft. The idea would be that when threading is selected, it shows all the options in the existing set up , and when the new feature is selected, it shows the feed amount using the different gear ratio settings. Just a thought, perhaps not practical or possible given my lack of experience with the current setup.

johnsonm, 2023-05-29:
That's why I pointed you to kwackers' #232 which has that feature.

Now, it doesn't have separate X and Y feed (which I need to be accurate; see #253) but if you don't have that, `STEPPER_RESOLUTION_FEED` on kwackers' version gives you what you want.

patl1, 2023-05-29:
I took a look at kwacker's post but didn't see where it had the additional parameter options. Probably don't understand how the system works yet till I mess with it a little.

johnsonm, 2023-05-29:
Sorry, actually, that feature is in the main branch here, I'm sorry. I *do* depend on other features in the kwackers branch, but `STEPPER_RESOLUTION_FEED` configuration is in @clough42's main branch. See this configuration file, which is where all configuration should be set:

https://github.com/clough42/electronic-leadscrew/blob/master/els-f280049c/Configuration.h

patl1, 2023-05-29:
Michael,
I understand the basic parameter setting that accepts entries to produce the correct ratio from the spindle to encoder to servo to the output threading shaft. What I was hoping to be able to do is have an additional screen that , in addition to the current tpi, mm pitch, and feedrate, that allows me to incorporate a separate ratio parameter in the firmware to use for the feedrate calculation since my feed shaft is turning at a 7:1 ratio to my threading shaft. Hope this makes sense

johnsonm, 2023-05-30:
`STEPPER_RESOLUTION_FEED` causes the feed to be done using a different ratio than threading. When you select a threading mode, it uses `STEPPER_RESOLUTION`, and when you select a feed mode, it uses `STEPPER_RESOLUTION_FEED`. This follows the normal use for your lathe, where threading is done with the lead screw, and feed is done with the feed bar.

It sounds like you want to sometimes use your lead screw to feed, and other times use your feed bar to feed. Since the reason to have a separate lead screw and feed bar is to reduce wear on the lead screw and thus keep it more accurate, can you describe the cases where you would choose to use your lead screw instead of your feed rod specifically for feed?

The feed mechanism is not as precise as the lead screw, so I assume you aren't actually thinking of threading using the feed bar.

I'm not aware that anyone else has actually wanted to use their lead screw for feeding when they have a feed bar specific to that purpose, which is why it would be hard to find someone who has already added UI for doing so.

patl1, 2023-05-30:
Michael, Thanks for your response. From your description it looks like it will do exactly what I want.  I'll take a closer look at the code so I better understand it and when I get my set up going, I can test it. Currently James is out of stock on the control panels so I will have to be patient. Glad you responded and it looks like the problem has been resolved. One thing that threw is that the "readme" file that downloaded with the code listed the solution as a future goal:

"Items on the short list for future development:
•Support for dual-shaft lathes that have different drive ratios for feeding and threading"

I figured it hadn't been addressed yet but looks like I was incorrect.

johnsonm, 2023-05-30:
It's been a while since James [merged a PR](https://github.com/clough42/electronic-leadscrew/pulls?q=is%3Apr+is%3Aclosed) — #96 was merged in October 2020.

His two items listed for possible future development in the readme ("Threading/feeding up to a hard shoulder with automatic stop" and "Automatic recall and resync with hybrid threads (e.g. metric threads on an imperial leadscrew)") already exist on kwackers' PR. I've offered to rebase that and take out the config changes from it to make it easy to merge without changing the base config, which would make it easier for everyone else to merge into their own branches, but only if James is at least willing to consider using that work. ☺

In default of that, I think that the kwackers branch is actually the active upstream for the firmware. ☺

patl1, 2023-05-31:
I am not really familiar with how branching and merging and rebasing works so a little guidance would be appreciated. Since the version I downloaded from cough42 github page (latest release) contains a readme file that says it is a future goal, should I assume that version of the firmware does not include the capability for separate feed ratio control? Do I need to go to someone elses git hub page (Kwackers?) to download a version that does include it? Thanks in advance for your help.

johnsonm, 2023-06-04:
You should be able to download from Kwackers' project. Then apply your changes. Note that he has different starting configuration.

https://github.com/kwackers/electronic-leadscrew

patl1, 2023-06-04:
Thanks Michael. I spent some time the other day trying to sor through the process and did exactly that, I only need a few more parts and the ELS should be up and running.

Thanks for your assistance.

Pat

 

https://github.com/kwackers/electronic-leadscrew

patl1, 2023-06-06:
Michael,

I have my set up nearly complete and hooked up the omron rotary encoder (E6B2_CWZ6C as recommended) to see if I could get an RPM signal. To my surprise, the RPM display remains at “0”. The servo is not yet connected as I need to make up the cable that runs to it. I hooked up the wiring to the Ti board and double checked the connections. Any idea(s) why I am not getting an rpm output? There is 5 V going to the encoder, I don’t know how to test the encoder output. I do have access to a digital oscilloscope and may kust hook it up to see what I see.

johnsonm, 2023-06-06:
The only time I saw that I had plugged the encoder into the wrong headers (there are two encoder ports on the dev board).

patl1, 2023-06-06:
I moved this discussion of the "no RPM" over to discussion #230

patl1, 2023-06-23:
The issues are resolved and the ELS is working very well.  I appreciate all the feedback I got to help get it there.

I re-opened the topic to offer the Fusion 360 files, pictures of the installation, gear selector ratios to use, parameter settings for the firmware, and pulleys/belts used on the Jet 1340 lathe to anyone interested. The F360 model contains the encoder bracket, 
 the stepper bracket (for the iSV57T-180 servo ), and a mock up of the lathe frame itself to allow locating the brackets precisely into the existing lathe casting cavities and are invisible when the lathe cover is closed.  The encoder bracket is a 3D printed part and the servo bracket is 1/4" steel plate weldment.