clough42/electronic-leadscrew #263: control panel fwd

freddygino, 2023-06-06:
hi guy finally got the els installed 2 problems if anyone can help....
1..on the control panel its on fwd but the lead screw is going oppisite

2...when i power off the control panel the lead screw is supposed to move free but it feels like the servo is still energized

tia

_Originally posted by @freddygino in https://github.com/clough42/electronic-leadscrew/issues/260#issuecomment-1577713815_

johnsonm, 2023-06-06:
You need to uncomment or comment (depending on what you have now)
```C
#define INVERT_DIRECTION_PIN true
```

What servo do you have? The iSVT7T, for example, doesn't have a enable line and so can't power on/off from the enable line, so doesn't have that feature.

freddygino, 2023-06-06:
yes have that servo same exact set up as james
[image]

johnsonm, 2023-06-06:
OK, without an enable line you can't disable the servo directly. You could add additional capability to control power to the servo if you wanted to and use the enable line for that.

But the direction depends on how you are linked mechanically, so you just have to invert the direction signal from what you had before, as I described.

freddygino, 2023-06-06:
> OK, without an enable line you can't disable the servo directly. You could add additional capability to control power to the servo if you wanted to and use the enable line for that.
>
> But the [...]

micheal thank you for your response....i got the fwd and rev done....so when the servo is not in use is it supposed to be energized when the control panel is off?