clough42/electronic-leadscrew #265: Power feed is rough?

javaduke, 2023-06-25:
I have a PM-1130 lathe with separate lead screws and power feed bars. I configured the values in the `Configuration.h` as follows:
```
#define STEPPER_MICROSTEPS 6
#define STEPPER_RESOLUTION 1000
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS
#define STEPPER_RESOLUTION_FEED 9375
```
I have 1:3 ratio from the servo to the gearbox and additional 1:2 ratio in the gearbox, so total is 1:6. Well, everything works, threads are fine, but I noticed that I'm getting somewhat rough finish when using power feed even at the .001" per rev, not to mention if I increase it to .005 or so, it looks more like a thread. And I noticed that the power bar does not rotate smoothly, instead it rotates in small pulses, so I'm wondering if this is what's causing a bad finish. Are my settings incorrect?
My servo and encoder are the exact models listed here: https://github.com/clough42/electronic-leadscrew/wiki/Stuff-You'll-Need#lathe-specific-parts (Integrated Nema 23 servo)

johnsonm, 2023-06-25:
Do you see the same thing when using power cross-feed?

If your feed bar is anything like mine (I have a G0709), it is below the oil line in the gear box and therefore sealed tightly. You might be right at the edge of what the servo can turn when feeding, so that the servo gets behind and catches up.

Did you tune the servo?

javaduke, 2023-06-25:
Yes, cross feed is a bit rough too. I was just turning some parts this morning and when feeding manually I get much better finish. 

As for tuning the servo, I just followed instructions in the James' video.

johnsonm, 2023-06-25:
I'm partway through the process of changing to at least 5.5:1 using change gears and mounting the servo on the banjo, but my gearbox is currently open as I've been cleaning it out, so I can't yet test that change.

It might be that 3:1 just isn't enough mechanical advantage for this servo driving a stiff feed bar. The fact that it is affecting crossfeed too means that it's not just stiffness moving the carriage. Here's what I've put together in case it gives you any useful ideas.

[image]

Right now I have a 3D printed gear on there for testing, but a set of gear cutters in the right module just arrived, so I'll be able to cut my own proper gear for it soon.

javaduke, 2023-06-25:
Hmm, I see, so the servo to shaft must be even higher. OK, thanks, I will experiment with it.

johnsonm, 2023-06-25:
It's at least possible. But also I don't know what you tried for stiffness and all the other parameters; @clough42 had a starting place but it's not necessarily perfect for every lathe. You might try the auto-tuning option with feed engaged and see how well it works? It's not crazy that you might be able to simply tune the servo to make it work.

javaduke, 2023-06-25:
Sorry, what's auto-tuning option? I don't remember seeing it in the software. Can you please point me to any instructions?

johnsonm, 2023-06-25:
Appendix A in the datasheet, "How to get the correct Load inertia ratio"

javaduke, 2023-06-26:
Ah, found it, thank you!

johnsonm, 2023-07-15:
I finally got to testing this with my lathe. At 6:1 on the input (11:66), I can't make it stall, and the default RatioOfInertia of 250 seems OK. That is, depending on settings and feed, cross-feed, or lead screw, I measure between about 125 and 375 when testing at the max RPM (3000), so it's in the middle of that range.

How many pulses per revolution did you configure the servo for?

johnsonm, 2023-08-27:
I have found that with the 11-tooth gear printed in PETG, I can program the servo to be stiffer without it singing. I'm no longer planning to cut a metal 11-tooth or 12-tooth gear at this time.

I put some comments into my branch that might help. See #270 for what I did.