clough42/electronic-leadscrew #266: Flickering screen

Thyerhazard, 2023-08-05:
Hey guys,

Really pulling my hair out with this, everything setup and running but I'm getting really bad flickering on the display and other fun weird stuff.

I'm on the latest 1.4.00 Firmware, at first I assumed noise was from the AC power coming in to power the lathe so to eliminate that I have temporarily ran the AC cable in the complete opposite side of the lathe no where near any of the ELS stuff or any other ELV stuff and even ran an extension cord across the shop to another outlet on a completely different circuit on another phase to what the lathe was on (Overkill I know, But I really wanted to eliminate that as the issue)

This did nothing to help, so my next step was replacing the display cable and shortening it as I saw James recommend in the event of flickering, It was originally around 1.6Meters long and un shielded, I've switched to a shieled cabled and managed to reduce the length to around 1 meter. and still the same flickering issue. 

I want to note that before changing the display cable to shielded the RPM readout was working and after changing it the RPM has disappeared, Also to note when the ELS was set to off with only the RPM displaying it did not flicker or have any issues only when the thing was on. ALSO (Sorry) When on the stepper motor would hold and not let the Leadscrew turn but it would not turn at all.

Sorry for the complete mess of test (Its 2:15am where I am and I have been bouncing my head against this all day, Really hoping Its a software issue or something you guys have seen before, Again sorry. and thanks for any help anyone can give me!

Thyerhazard, 2023-08-06:
https://imgur.com/a/uz6Z58s

Here is a link to a short video, showing the Screen issues and the Stepper motor locking out the leadscrew when the ELS is on.

Also while recording this video I had TOO FAST come on on the display which I had not seen before so I'm glad I caught it on video, I'm really at a loss for what is happening, I've gone though the trouble shooting page, and continuity tested everything I can think off, 

I seems like the data is corrupt somehow, I really hope someone has had a similar issue and managed to find an answer.

Thanks guys!

Thyerhazard, 2023-08-06:
https://imgur.com/a/C0FQIQW

*Update* 

Ok I've ripped it all out again (note to self never do cable management before testing that's 3 hours Ill never get back and a lot of wasted zip ties)
 
I have it back on the bench and its still flickering in the same exact way, So now I know for sure there is no noise effecting it. something else must be going on, I'm not a complete novice with electronics (I am an licenced electrician) but I'm not an expert in electronics by any means and do not have an oscilloscope, best I can do is look over everything with my microscope and hopefully find a spec of solder crossing pads.

Will update my findings as they happen,

Thyerhazard, 2023-08-06:
Can't see anything wrong, Guessing I'm missing something or have a dud board, but without test gear have no idea what is or could be dead. Hopefully there is stock soon as ATM all I can do is but another Boost board and Display and try again.

Chucksbp, 2023-08-06:
Clough 42 is developing a new display board because of failures on the boards. that being said in the interim I would order a couple boards fro ebay or Amazon and see if you can get a good one. I had a bad board right from the start.  on the second attempt i didnt change the plug t, i just bent the 5 lugs at a 90 degree angle and pluged directly in, changing the wire position a bit.  as far as the no RPM reding is your stepper motor still operating or did it also stop. If its not opereating I would say that your incoder is toast! if you read down below you will find this is happening with the ebay encoders. Teres a much better one from amazon this also happened to me about three hours in on my build.I used the Amazon encoder and reprogramed the board for the second input and its been working properly for over a year. well worth the efforts after you get things working!

Thyerhazard, 2023-08-07:
Yeah, I'm going to have to pick up a new Display board and booster when they become available, Unfortunately for me with prices of shipping to Australia its not worth ordering a bunch from china and hoping I get a good one, better off waiting for a quality one from James directly.

Also my encoder is fine, When I bench tested it my RPM readout came back, I think my hobbled together test cable was just a bit 'how ya goin' haha Thanks for the reply mate!

Thyerhazard, 2023-08-26:
[image]

Update, So I got it working! I ordered 8 new LED and KEYBOARDs and 7 were DOA. But I got it working in the end I also had some continuity issues with the Green Clough42 board that would be there sometimes and gone the next so I soldered in some jumper cables to be safe and so far so good everything SEEMS to be alive again.

Chucksbp, 2023-08-26:
im glad you have it working!  If you dont want to disturb the display by changing to the angle pins on the other side of the board., I just bent the existing pins over leaving enough space for the plastic plug. If I rember orrectly i filed the emd of the plastic plug down a bit so that it wemt on a bit further for clearance inside the box. It was a snug fit but has workd just fine.