clough42/electronic-leadscrew #267: Modules and diametral pitches

johnsonm, 2023-08-05:
My G0709 lathe has configuration for cutting both metric module and imperial diametral pitches, for cutting worms that are compatible with metric and imperial gears, respectively.

I don't see that anyone has implemented support for these pi-based pitches. I'm not even sure what the UI could be on the default control panel, honestly. Has anyone here thought about this?

johnsonm, 2023-08-06:
I guess one UI would be to build on top of #232 and add a WORM setting which honors the mm/inch feed setting, but multiplies the numerator by pi.

blackfister, 2023-11-15:
Hi there, I have made worms by adding “kwakers”. It gives you any sizing of thread.  You can use either metric or imperial. For metric module to imperial diametral pitch, you can use online convertors

johnsonm, 2023-11-16:
Yes, that's #232 and I've based my own work on his work generally. I've even offered to rebase it, preserving the original configuration to make it easier for @clough42 to merge it, but didn't hear back. So at this point, as far as I can tell, kwackers is the current upstream. :grin: 

My real question was what the UI would best be on this limited control panel if implementing module and diametral pitches directly, without conversion.