clough42/electronic-leadscrew #268: Carriage stop/emergency switch?

javaduke, 2023-08-07:
I'm thinking about making a custom carriage stop with a switch that will cut the servo power off when engaged by the carriage. 
This can essentially be a poor man's thread/cut to shoulder feature - I would simply set a carriage stop at the predefined location and then the cutter would stop at this location even though the power feed or lead screw is still engaged.

Wondering if anyone can think of any possible downsides of this approach? And what would be the best place in the circuits to include this switch? I can simply cut the power line to the servo and insert the switch there, or is there a better place?

kwackers, 2023-08-08:
The problem with that is you'll lose sync between the carriage position and the spindle since the carriage will stop but the spindle wont...

Have a look at the thread to shoulder code I added - it's cheaper than a switch... ;)
[](https://github.com/kwackers/electronic-leadscrew)

javaduke, 2023-08-10:
That's more or less what I have in mind, emergency switch + turning/boring to shoulder, my only question is where would be a good place for a switch on this diagram 
[image]
Would it be the red PUL wire? Or somewhere in the power supply line (i.e. cutting the power to the motor?

tomvikse, 2023-08-10:
.. what if you cut the encoder signal?

Chucksbp, 2023-08-10:
As  I mentioned I cut the enable wire going to the servo, this stops the motor quickly. It will stop every time within .002 of the shoulder. I ran my micro switch and an emergency stop in series. makes for an easy stop. Im not sure about the encoder although i would think if you cut the 5 volt signal wire to the encoder it would also stop the servo.

javaduke, 2023-08-10:
Which one is the enable wire on that diagram? I see PUL, DIR and  ALM, which one I should cut?

tomvikse, 2023-08-10:
The servo in the picture dont have enable signal.. so you cant use that..

Chucksbp, 2023-08-10:
Direction wire wont do it so try the pull wire, just disconnect it at the els board and see what happens. If it doesnt stop it try the red + wire that should stop the servo.