clough42/electronic-leadscrew #274: 2 years of on and off trying to get this to work

markcurington, 2023-10-19:
With these parameters .040 feed just barely moves. I can up stepper microsteps  alot and get things moving but even setting microsteps to 36 and changing resolution to 400 in the program and on the driver won't drive the motor fast enough. Something is missing and i can't figure it out. I got an RPM meter from work a year or so ago  and the speed matches the display. I bought one that should arrive this week because i don't work at the same place anymore. This is my last day of normal hours for a few weeks I start working 12's for the next 2 weeks. The plant is shutting down for the annual maintenance cycle and lucky me I'm the only electrician at the plant. Can someone help me understand what I'm doing wrong so I can fix this problem when I get done with this outage. I really want to learn how to thread on my lathe.

 I have a 8 TPI leadscrew.
1024 encoder parameter set to 4096
steeper microsteps 6                             
stepper resolution set to 1000
10 tooth gear on the stepper motor 
60 tooth gear on the leadscrew
spindle and encoder are 1:1
[image]
[image]

dsgroninger, 2023-10-20:
Mark,

Please see my writeup on my implementation of the project.  It sounds like we have similar lathes, and I included my changes to configuration.h plus the logic that I used in setting up my encoder.  If your controller is correctly reading the spindle speed, and if the stepper motor moves the lead screw at all, then you're 98% of the way there.
[Clough42 Electronic Leadscrew.pdf](https://github.com/clough42/electronic-leadscrew/files/13048582/Clough42.Electronic.Leadscrew.pdf)

Good luck,

Dan

markcurington, 2023-10-20:
Thanks Dan. I'll read up. I verified today with my new RPM meter that the spindle speed matches the display speed.

markcurington, 2023-10-21:
So i measured my saddle travel tonight and it moved around .125.  According to your writeup you have a 3 to 1 so 3x3.333=10. I have a 6:1 so i did 6x3.333=20. I got errors in my table.ccp and it won't compile. I'm going to try a 20 tooth on the stepper motor and go with 3:1 and see if that fixes things tomorrow if i get off work at a reasonable time.
[image]

markcurington, 2023-10-22:
Okay so I found where I messed up last night. Where James code says "#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS" Dan's writeup said to change this but you I had to remove STEPPER_MICROSTEPS. It will not compile with #define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 20 because that causes errors in the tables.ccp. In order to get it to work I had to delete STEPPER_MICROSTEPS then define with 20. It now reads #define STEPPER_MICROSTEPS_FEED 20 and compiles.

markcurington, 2023-10-22:
Well that didn't work. Stepper motor still not turning fast enough.

dsgroninger, 2023-10-23:
Check the settings on your stepper motor controller.  You may need a particular serial cable from the maker of the controller.   You need to be certain of how many pulses, or “tics” it takes to make the motor move through one revolution.  On my controller that was set with DIP switches, and it turned out that hi and lo weren’t obvious on the switch and it was set up backwards from the factory.  It could be that the math in the software is all fine, but the motor controller is set up for some wrong number of steps or micro steps per revolution. DanSent from my iPhone