clough42/electronic-leadscrew #277: Launchpad failure

mattdifelice, 2023-11-29:
Hello,

I am about to be going onto my 3rd launchpad after two have failed for this project. For the first failure, I have no explanation, I had it set up for a dry run (just all of the electronics connected) and it worked. The next day I went to plug it in, and the board was completely dead. For the second board, the only thing I can think is that the high frequency from my TIG welder killed it. The encoder also appears to have failed, but I have no way of telling if it was from the first or second time. I used shielded cable on all connections, grounded to the enclosure. Just trying to see if anyone has suggestions or has had similar issues. Thanks!

Chucksbp, 2023-11-29:
If you look below in older issues we had some failures with the import encoders taking out the input of the Launch board. there is a better enoder on Amazon that seems to have solved that problem at least for me and a couple other gentelman. i was able to salvage my board buy using the second set of inputs on the launch board. this is the encodr we had good luck with. https://www.amazon.com/gp/product/B08RS6M32J/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1 

 what were you Tig welding that would cause the boarad to have issues. 
You may also want to be carefull about your grounds Clough 42 has some precautions mentioned. It has been quite a while since i built mine so i dont remember the exact cautions. grounding shields are fine but not other common grounds on the board i believe. 
  The boards are Touchy im sure, they are experimental boards from my understanding. luckily My ELS has beeen working well  since the original failure, although I mad a second complete electronics set up and keep a 3rd sare board just in case. they were hard to come by during covid.

mattdifelice, 2023-11-29:
Thanks for the reply. As far as the TIG welding goes, it wasn't anything associated with the ELS and was about 10 feet away. I just wasn't sure if the high frequency noise was enough to do any damage. I will also get that other encoder ordered to be safe. How did you go about using the second set of inputs? Is it something in the code or a matter of changing switches? I am not extremely tech savvy but I would like to try to salvage the board before making another attempt with a new one.

Chucksbp, 2023-11-29:
Your welcome! I dont think any eletrical noise would be an issue unless in direct contact with the board or grounds. Although there is always a slight chance of statuic discharge when handeling the boards. 
Yes there are two sets of terminals on the board, one above the other. In code composer the choices are on line 102 and 103. Simplly remove the hash marks from line 103 and replace them in line 102 and reprogram your board.  

  I dont know how significant your board failure  was or is. On mine and a few others the ELS powered up normally but would not show the Tachometer reading nor operate the stepper motor. there were quite a few instances in 2021 with what we felt were a bad bunch of encoders. no one tried the old encoder on the second set of inputs. as i mentioned at that time Launch boards were almost impossable to frind with covid and the chip shortage. 
 You may also want to check your boards connections for any shorts to ground other than the actual ground pins. 
  it may make things eaasier, in my case i just remove the board from my case, reinstall the jumpers on the board and use just the supplied cable to program the board. that way you dont need an external power supply or yhe Clough42 jumper board to program the Launch board after programing remove the jumpers and reinstall.

hr2burn, 2023-11-29:
FWIW, when I AC weld at high frequency, my cable box (inside the house, 30-40’ away) goes to static and I need to unplug reboot it. If something is sensitive, I guess anything is possible. My Clough ELS is in the shop with welder, but I’ve never had it on at the same time as welding. It’s on a power strip turned off essentially disconnecting it from the house’s electrical system.

Chucksbp, 2023-11-29:
My guess is that the cable box is going crazy from feedback on the AC line not from EMS. And again it isn't ruining the cable box just disturbing it. But then again who knows, I would be sure not to have the ELS or for that matter a DRO on when welding.  I did have DRO problem on my lathe that was caused by a new style electronic florescent ballast. The DRO measurement would jump occasionally at random intervals by large amounts. I changed the light to an old magnetic style ballast and never had a jump after that. That I'm sure was an EMS issue. Sent from my Galaxy