clough42/electronic-leadscrew #278: Spindle position display option

R-Ronald, 2023-12-29:
Greetings. I've just completed my ELS install today and have it working on my PM-1130v lathe. Thanks to all who have made this project possible!

There is another piece of data the the ELS controller has that would be helpful. The ELS controller knows the spindle position to a resolution of a tiny bit better than a 10th of a degree. Displaying this position would allow the lathe operator to rotate the spindle and workpiece to a particular position. For example, I could scribe the side of a pulley busing with two marks at exactly 90 degrees for subsequent drilling of set screw holes.

Someone who wanted to use this capability in its full glory might want to add a spindle lock and/or some mechanism to rotate the spindle more accurately or quickly than can be done by hand. But even w/o anything other than a toolbit to scratch a mark, this feature would allow using the spindle to rotate and mark workpieces to a 1/10th degree accuracy. A big win, IMHO.

There are only two operator inputs needed. 

1) There needs to be a way to toggle between RPM and Spindle Position. 
    - This could be a configuration switch in the code (I'd probably chose this over RPM since I already have a RPM gauge on the lathe).
    - It could be something shown when the ELS is in the "powered off" state (seems far more useful than just showing a zero :-). 
    - It could be a combo of buttons (multiple buttons do get confusing though)
    - It could be a button press before and during power up

2. It would be nice to be able to zero the angular position readout. This would allow display of angle just like a DRO. One can set the zero position and then rotate relative to that.

I'd vote to show the rotational angle measurement anytime the ELS is "off". And to allow the the set button in this mode to zero the current angle, so that relative movement doesn't require math on the operator's part.

Is there any interest in this?

RR

tomvikse, 2023-12-29:
Here you go
https://github.com/clough42/electronic-leadscrew/issues/231

johnsonm, 2023-12-29:
Seconding @tomvikse 's answer. I think that #231 is currently effectively the primary upstream for this firmware; it implements features that @clough42 anticipated potentially adding, and James hasn't been merging PRs generally into his branch. (I've offered to rebase it to make it easier to merge into mainstream, preserving the default configuration, but James hasn't taken me up on the offer.)

I used the angle display to accurately tune my feed rate with a dial indicator for my lathe; I have both a lead screw and a feed bar, and I didn't characterize the gear ratios in my apron, so I just took the backlash out, then measured error while turning the chuck by hand. Super handy, and since I don't have a dividing head for my spindle, probably the most accurate thing I could do.

However, #231 doesn't have your idea of zeroing out the angle, I think. That would be a potentially useful addition! ☺

R-Ronald, 2023-12-29:
I'm not sure I'm ready to digest all the changes in the #231 firmware, but it sounds a good portion of the RPM code that I want is in there. Having the angular position when the ELS is "off" and having it operate normally when on is still of interest. Since all the hard coding is probably done and available, I might overcome my coding aversion and try implementing it at some point when I need a dividing head. But, I need to ponder for a bit.

FYI, you don't really need the angular position display to calibrate for a feed vs a thread operation. I marked the spindle with a sharpy and put a corresponding mark on the sheet metal above that mark. While rotating the spindle by hand and lining it up by eye is prone to minor errors they can be reduced to almost nothing by rotating the spindle 10x rather than 1x to get the distance per revolution. And, to top it off, I don't see any need for great precision on feed rates. At least for me, .11 mm/rev and .13 mm/rev would be so close to .12mm/rev that I wouldn't know the difference. Not saying that the precise approach isn't better....just that it's akin to polishing a cannonball. 

Strangely enough, my calculated "microsteps" for my feed bar is 4.71 (compared to 3 for the thread microsteps). Not sure what gearing is in place to get that weird ratio, but I can't argue with success.

R-Ronald, 2023-12-30:
Angular position via a display seems useless if the lathe is running. So I suspect that a reset of angular position is only useful (and therefore only desired) when the lathe is not running. In the "powered off" state, I was hoping that the "set" button might be available.

johnsonm, 2023-12-30:
That seems smart.

I'd think that if entering the position mode reset the angle, it would also reset threading start, though. :thinking: 

I wonder whether having a separate "display angle" offset and resetting that offset such that the display is zero when entering position mode would be the best way? Maybe the `UserInterface` could hold and apply the offset, leaving the angle in the `Encoder` alone?

R-Ronald, 2023-12-30:
Don't you lose threading start when you go to position mode regardless? You'd have to leave the "Thread to Shoulder" mode to select a new mode. And if you wanted to go back to threading, you'd need to select a new mode?

The display angle can definitely be different than the "encoder" value. To zero (**at the start of position mode**), it should read and store the initial encoder value (effectively storing the zero position). Then the display routine can display the difference of the current encoder position to the saved encoder position. Converted to degrees and tenths. Always a positive value that is never more than 360. I'm sure there's a modulo operation and an ABS operation in the line...or something like that.

I may get to load and try the #231 code this afternoon.

johnsonm, 2023-12-30:
You are right, I was over-thinking it.

In any case, the angle is kept in a `Uint16` integer that holds tenths of a degree, so you can use the `%` modulo operator. Just add a positive offset and then `% 3600`

R-Ronald, 2023-12-31:
I didn't get to try it, but I am looking at the code. In Encoder.cpp  of the #231 branch, the last few lines read:

Uint16 Encoder :: getSpindleAngle(void)
{ spindleAngle = (getPosition() % ENCODER_RESOLUTION * 3600) / ENCODER_RESOLUTION;
    return spindleAngle; }

There are are least two statements in that file initializing spindle angle to zero.

I made my config changes and am trying to build, but the stupid Code Composer hammer icon is greyed out. I can right click on main.cpp and there's a build option, but it doesn't do anything......sigh

R-Ronald, 2023-12-31:
If I click on the lower level item in the project hierarchy it works. But I still hate Eclipse.

R-Ronald, 2023-12-31:
I got the #231 code built and loaded.

Unfortunately, the angle readout currently doesn't get reset when the "position mode" is entered. It's hard to hit zero, but it is possible with a tapity-tap-tap action on the spindle. 

I'm going to think a bit about next steps. I like the idea of the other features, but it does add significant complexity to the operator interface. The limits of the display and the buttons are the biggest cause of this.

One other thought nagging at me is that there is no emergency stop for the feed in Thread to Shoulder. The thread to Shoulder function moves the carriage based on a button press and stopping that would not be easy using the "power" key on the Clough42 display.

johnsonm, 2024-01-01:
I'm just glancing at the code, and haven't thought through it, but I'm expecting that resetting the encoder when exiting the menu if showing the angle has been selected would work. Something vaguely like this:

```c++
    case kShowPosition+2:   // run loop
        showAngle = !showAngle;
        if (showAngle) {
            encoder->.reset();
        }
        clearMessage();
        this->menuState = kQuitMenu;
        break;
```

For emergency stop, I think that a real emergency stop never depends on logic in code. I power my ELS through a power supply behind a contactor that is piggy-backed off the main lathe contactor, so the lathe emergency stop stops the ELS along with the lathe motor.

R-Ronald, 2024-01-02:
I tried building that code, but I don't think there's a **encoder ->.reset** in place at this point.

But I agree....that's where it needs to go.

johnsonm, 2024-01-02:
I did say "vaguely" — but I've edited out the errant character.

If it works, I don't mind creating the PR for it.

R-Ronald, 2024-01-02:
No dice. for the test. :-(

The change does reset the displayed angle to zero. But there's an artifact. At the moment of the hitting "set" to select Position mode, there's a sudden movement of the leadscrew (or hex bar). I didn't spend much time characterizing the movement...but it was not subtle. On the one occurrence where I opened the gear cover to watch, it rotated the input shaft to the gearbox ~180 degrees. To me, it's unacceptable (and a bit scary) to see unintended motion.

While I suspect it is possible to code some sort of safe reset, it now definitely seems like a better idea to record the initial reported angle upon entering position mode. Then modify the angle display routine to offset the current angle by the initial angle each time the angle is displayed. This seems like a very safe option.

johnsonm, 2024-01-02:
OK, back to my original idea of storing an offset in the `UserInterface`...

kwackers/electronic-leadscrew#4 is my untested attempt to implement that. I'm away from my lathe at the moment, so like my first idea it's untested. But if you test it and it works, I can take it out of draft and ask @kwackers to consider merging it.

kwackers, 2024-01-02:
Hello guys.
I just got the notification that someone had posted something, so a quick scan of the posts and a quick scan of the code to see what's happening.
Bearing in mind the code falls into the category of "code I wrote more than a week ago" I'm just wondering if something like "press and hold set to reset" wouldn't be better?

I did at some point consider allowing the display to show the physical position of the saddle too - "press and hold" would also work there (should it ever get implemented).

My main thoughts are: 
Press and hold is a common way to zero something.
It's use doesn't preclude coming out of the angle mode and going back in without losing the position.

Thoughts?

johnsonm, 2024-01-03:
That would delay the menu until releasing SET, but yeah that could work. No promises but I might take a crack at it...

R-Ronald, 2024-01-03:
Press and Hold set to reset the displayed angle sounds great with one caveat. "Resetting the displayed angle" should not make the stepper or servo seek a new position. 

If we need to create a separate angle variable just for the display in order to avoid upsetting the other functionality of the ELS, Michael Johnson's has graciously offered a starting point for the code.

johnsonm, 2024-01-03:
@R-Ronald I don't know whether what I just force-pushed to the branch will even compile, since I'm not even near the system that has the development environment installed let alone near the lathe, but if you'd like to test it, let me know both of any stupid mistakes like the extra `.` as well as whether it works...

R-Ronald, 2024-01-03:
>   (Issue #278)
> Yeah, I won't reset() the encoder — your earlier test showed that would be a bad idea. My original idea three days ago was right after all. 😁
> Restating to make sure I understand right: The proposal here is that instead of resetting the display-only offset when going into angle display mode, we should reset the display-only offset only on a long-press of SET, only if it is already in angle display mode. (Later, it might reset the display-only linear carriage offset, only if it is already in the proposed new carriage offset mode.)

I had a different idea about how things should work.I thought that Position Mode could be selected w/o forcing the angle to zero by a short set button press or it could be selected along with forcing the initial angle to zero by a long set button press. Both options enter the mode and are doing a similar function.Otherwise, once in Position mode the set button press either exits the mode (when short) or resets the angle (when long). The two options use one key, but are totally different functions.Sent from my Galaxy

kwackers, 2024-01-03:
Quick look at the code (whilst waiting for stuff to build here).
In ControlPanel :: isStable the code waits for the keypresses to settle, this function could be expanded to allow the setting of a ControlPanel::isHeld value if the button is held beyond some predetermined time.
I'd change the line if( this->stableCount < MIN_CONSECUTIVE_READS ) to check against HELD_TIME instead and set 'isHeld' if that occurs (as well as clearing it when the statement isn't true).

In UserInterface :: mainLoop that needs to enter a state when the set button is pressed instead of immediately calling the menu.
e.g. "WaitForSetReleaseOrHeld" - that sort of thing.
Then each loop it'll check that state and make sure the set button is still held, if not then call the menu, if it is then wait until Control::isHeld becomes true at which point you can clear the angle.
Once the angle is cleared the above state needs to advance so that it knows not to call the menu when the button is finally released.

I can easily add this functionality as some point if you wish - just need a bit of time to grab the controller off the lathe for testing.

FWIW
I think the current system of having the 'mode' for the display permanent until deliberately changed is preferable to 'cancelling' it by pressing the Set button and I can't say I'd be a fan of having two mechanisms of entering angle mode one of which clears.
Simply clearing it by pressing and holding Set seems far simpler and less confusing to me.

johnsonm, 2024-01-03:
I think I'll want to actually trace through this when I get back to my lathe to make sure I understand the state machines in the `UserInterface` correctly.

kwackers, 2024-01-03:
I think the code to read the keypad should look something like this:
(With suitable value for HOLD_TIME)

bool ControlPanel :: isStable(KEY_REG testKeys) {
    // don't trust any read key state until we've seen it multiple times consecutively (noise filter)
    if( testKeys.all != stableKeys.all )
    {
        this->stableKeys = testKeys;
        this->stableCount = 1;
        this->isHeld = false;
    }
    else
    {
        if (this->stableCount < HOLD_TIME)
        {
            this->stableCount++;
        }
        else
        {
            this->isHeld = true;
        }
    }

    return this->stableCount >= MIN_CONSECUTIVE_READS;
}

johnsonm, 2024-01-03:
OK, doing it in the `ControlPanel` instead. I wondered which way would make more sense. Then I can throw away my current attempt and start over once I have the chance to debug at my lathe... ☺

kwackers, 2024-01-03:
And the code for the keypress in UserInterface::mainLoop like this:
(With setHeldState added and the function clearAngle defined)

`            if( keys.bit.SET )
            {
                // set pressed
                this->setHeldState = 1;

                // has it been held long enough?
                if (this->controlPanel.isHeld)
                {
                    // clear angle and prevent menu appearing when released
                    clearAngle();
                    this->setHeldState = 2;
                }
            }
            else
            {
                // set has been pressed and released
                if (this->setHeldState == 1)
                    beginMenu();
                this->setHeldState = 0;
            }
`

kwackers, 2024-01-03:
Of course I haven't tried any of the above... ;)

R-Ronald, 2024-01-04:
So, I was busy today...actually running my lathe. I'd 3d printed a gear to drive the ELS encoder, and was machining an aluminum replacement (3d part is fine, but I have a lathe and feel compelled to use it :-) 

The ELS was great. Makes everything much quieter. And quieter seems less stressful too. 

But, I can see a couple more operations that could be improved by ELS modes.  So I'll just throw these out as an idea to see what you think.

1. First, I was boring out an 8mm bore pulley to 48mm.  I'm uncertain of my workholding, so I didn't want to take big cuts. Was doing .5 to .7 mm off the diameter for each pass. Thus, I had to do a lot of cuts (~70?). It would be nice to have something like the Thread to Shoulder for normal cuts (I was doing an internal bore, but external would be the same). 

Set the Start, Set the end, and go. For the feeding I was doing the spindle didn't even need to stop. Feed from start to end, stop feed, reverse feed back to start and repeat. It could be done w/o any further operator panel interaction. You could have a pause of a second or two at each direction change to allow time to move the cross slide....or not (you actually already have a mechanical pause during reversals due to backlash. 

I'd call it PingPong mode. One could cut in both directions (although I would have been chicken to do this since it would be trying to pull the weakly held part from the chuck)....or the reverse could be at a faster travel rate. 

Turning off the spindle could be used to signal a stop for the PingPong action.

2. Once I completed the bore, I had to broach a keyway. Lots of hard work running the carriage back and forth w/o the spindle even turning. Again, there's a start point and a stop point. Ping pong between the two. On this, you'd almost certainly not want to stop at the endpoints.

The broach operation is a bit scary. If you attempt to cut too deeply when moving the carriage by hand for broaching, it just klunks to a stop. Not sure what the servo would do if it was pushing and was stopped (trips off?) and whether the resistance makes for a hard crash or just stall w/o breaking anything.

RR

johnsonm, 2024-01-04:
I found that a gear printed in PETG actually allowed me to increase stiffness without feedback (as indicated in my Hobby-Machinist comment in the thread we're both in there: https://www.hobby-machinist.com/threads/has-anyone-installed-clough42s-electronic-leadscrew-on-a-pm-machine.95057/post-1094101  — So even though I bought 1.5 module cutters expressly to cut a brass gear, I changed my mind and haven't used them.

On my lathe, the feed bar has a clutch. If that tripped, it would of course invalidate position. (I've never tripped it, as far as I recall.)

I'd almost certainly use a "turn to shoulder" mode myself. I've thought already about adding it. If I do I'll put up a PR for it. But I'd suggest tracking that as a separate feature suggestion. I would definitely cut in one direction and just use fast reverse, which the ELS makes easy.

I definitely wouldn't use it for broaching. If the servo trips overcurrent, you have to power-cycle it. And broaching is a higher force operation, more likely to do damage.

kwackers, 2024-01-04:
Thread to shoulder together with fine feed works well for machining stuff but an automatic return would be nice (I've never owned a lathe that didn't take a shaving off returning).
So some sort of ping pong mode is worth a punt, needs playing with a bit since often you want to move the carriage out of the way to take some measurements (or at least I do) so would need some mechanism to allow that without losing the setup.

R-Ronald, 2024-01-04:
Michael-The gear I was working on was just to drive the encoder. The plastic one I replaced one was ABS. I used to like PETG, but ABS with an enclosure and a PEI plate is amazing. Parts never pull off during printing and they are completely loose after the plate cools off. On PEI, PETG sticks too well.

I'm no longer convinced that "stiffness" matters for our usage of the servo. I know that Clough42 was trying to increase it, but never saw any evidence that it made anything better when increased.....or worse when decreased. And I have seen people drop back to a setting of 9 (with no mention that this causes any other problems).

Much as I'd like "power broaching", your input that it's a bad idea is pretty convincing. 

Steve- For a Cut to Shoulder (or ping pong), "Resume" feature, I'd require three things.

1. Spindle is spinning again.
2. Some sort of prompt and ack at the panel. E.g., panel displays RESUME CUT
3. Operator hits a key to agree.

I guess that carriage manual motion by the operator to gain access would be a problem. Either the operator needs to end the manual moves by returning to the same place before resuming (set the DRO to zero at the start location to allow for an easy resume).....or he needs to use a power feed feature to move the carriage for access (while keeping the system knowledgeable about carriage location.)

kwackers, 2024-01-04:
So the way I'd see the ping pong mode working is you select it and set the start and end just like thread-to-shoulder.
I'm thinking I'd also let you select the direction of cut - in/out/both
Where both would be a true ping pong and cuts in both directions.

So it performs the cut and ends up with the saddle back at the start and some message telling you to press a button to repeat.
You then press said button to repeat the cut (assuming you've tweaked the depth for the next).

If you stop the spindle then you can press the +/- buttons to manually move the saddle for measuring etc.
When you restart the spindle and returns to displaying the message but now when the button is pressed it'll rapidly advance the saddle to the start position before repeating the cut.

I think the above gives you the ability to quickly move the saddle out of the way for measuring etc.

Possibly an even better refinement is to allow the setting of a "saddle park" position as part of the set up.
Then when the cut is made simply stopping the lathe will automagically move the saddle to the park position and restarting the lathe would return it to the start position.

R-Ronald, 2024-01-04:
I really like the saddle park idea......except that the act of turning off the lathe spindle could then be the trigger to make the carriage rapid towards doom (if someone forgot this feature, or left something in the way, or set the wrong park location, etc.)

I think it's much safer to always use the +/- buttons to move the carriage where you want it. If that makes it rapid towards doom, you'll probably be able pull your finger off the button much faster than you could shutdown the ELS.

kwackers, 2024-01-04:
So initially there is no safe position. 
At the end of the 1st pass when you stop the spindle it would remain where it was.

If you then shift the saddle position using +/- I think it's safe to assume that wherever you leave it is a "safe" park position.
Then you take another cut and it will again stop at the start, if you then stop the lathe it'll moves the carriage to the position you set above.

If you stop the lathe before it returns to the start position then nothing happens, in fact until you allow it to finish the cut (or exit the mode) then you won't even get the chance to tell it to repeat.

It's possible to make this configurable in the config file so you can disable it so TBH it's no biggie (but it would work for me).

R-Ronald, 2024-01-04:
Optional Auto Park. You'll have the Tesla of lathes :-)