clough42/electronic-leadscrew #279: SOS Reset sometimes

Olazas, 2024-01-07:
Hello!!!

Ive detected some time the controlPanel unit make reset.....  At the start show the initial title reset and show again the title..... and sometimes when I change from IN to MM,,, sometimes reset again, and sometime also I change thread to feed and reset too.

All works fine,,, but What will be that if we take in mind that 5v power supply is ok??

Ive got ELS 1.4.00

Thanks a lot!!!

Turner75, 2024-01-07:
I would suggest you have an intermittent connection problem on your display
board or interface board, most likely, the display board. The fact that it
sometimes occurs when you press the buttons would be an indication of a
loose connection, dry joint or unsoldered joint being disturbed as you
put some slight pressure on the display board by pressing on the button.
Just a thought  based on decades of fault location carried out personally.

Olazas, 2024-01-08:
Thanks a lot for your message Turner75.... Ive checked all point of ControlPanel and ELS, also wires,  and everything is ok

I,ve got ELS Boost V2.1 from 9/2019 and the Control Panel also from that date

Olazas, 2024-01-08:
Thanks a lot for yout reply with thats great information!!

Sorry, I didnt explain so well... Its a new instalation but the componets are from 2019

Ive tried to supply 5v outside of the box.....  you are right!!!!!  That way its no reset....

Please could you tell me what kind of filter you use.... I would like to keep all in the box,,, would be that possible??

I attach a pic of my box

Thanks a lot!!!

[image]

Chucksbp, 2024-01-08:
If the display is from 2019 i would highly suspect the display board. I did mine in early 2020 and went through several boards before i got a good one. fro my understanding Clough42 has begun having his owne disoay boards produced. it may be worth while to get one of his new boards. Lack of qualty boards was the reasonJames started producing his board.