clough42/electronic-leadscrew #280: Key&LED only works with USB power to launchpad

Bahr01, 2024-01-11:
I'm having an issue which has continued over 2 launchpads and 2 ELS boards from Clough42:

After flashing board, ELS will function perfectly when powered through USB on Launchpad with jumpers installed.
When using 5v power supply (jumpers removed), it just has a single red light on the board and nothing on the LCD.
I've verified 5v at the +5V pin and +3V at the correct pin.

This has me a bit stumped. I'd assumed it was a faulty board, but after replacing both, it's the same.
[image]

Chucksbp, 2024-01-18:
I would try a differant power supply and also double check the switch positions. some times you have to flip those back and fort a few times to eliminate the goo from the adhiseve cover. lastly check your continuity tp the board for the power  pins.

Bahr01, 2024-02-01:
Turns out to be dud LED&Keys, I just had 3 of them and I think what's happening is it works on my desk when I flash the board and test it, then my Meanwell 5v DIN psu I'm using to power the ELS board somehow fries the LED&Key, then I think it only works with USB.

I tested these back at my desk and they don't work at all, then found an LED&KEY I hadn't put in service and it works fine. I just need to find out why my meanwell psu is killing these boards, it's 5.0v on the money output, I'll have to put a scope on and check ripple.

Chucksbp, 2024-02-01:
This is the pow supply I have used on several electronic power supplies and it has worked well on all including the ELS.    https://www.amazon.com/gp/product/B01ALGGK2O/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

Bahr01, 2024-02-17:
So I have a final resolution; two things were wrong; I had a bad led&key (qc is shite on those). 2nd problem, my clough42 interface board was bad and not powering the launchpad properly. My workaround was to actually leave the jumpers connected on the launchpad and now it works perfectly. Still some flicker which I'm chasing down, but still usable.
The interface boards seem rather delicate, I have another in case it all goes south.

closed 2024-02-17 by Bahr01