clough42/electronic-leadscrew #281: Encoder requires ELS FWD/REV button push to change stepper rotation.

MMCCLENAHEN, 2024-01-29:
Has anyone else had the issue where the stepper motor only changes rotation direction when you change  FWD/REV on the ELS control panel? So when you rotate the  encoder clockwise the stepper motor rotates clockwise,  when you rotate the encoder counterclockwise the stepper motor will only rotate counterclockwise when the FWD / REV button is pushed on the ELS control panel.
Thanks in advance.

Chucksbp, 2024-01-30:
I havent had this issue, but I have had an encoder fail as quite a few other have had also. My thought would be to carefully check the switches on the launch board for proper positioning, i would also flick them back and forth a few times to make sure they are moving the entire way. then id check your pin crimps at the connector for continuity.  possibly one of the phases is not working?? you may also want to reprogram your Launch board to use the second imput. If all else fails id try a differant encoder. I have found a much better quality encoder on Amazon. https://www.amazon.com/gp/product/B08RS6M32J/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

dsgroninger, 2024-01-30:
Check the wiring from the encoder to the TI board as well.  If only one phase of the encoder is connected or operating, the TI board will still count tics, but it won't know which direction the encoder is turning.  Quadrature encoders take power and ground and output square wave signals out of phase with each other.  The TI board counts square wave pulses, and if phase A is leading B it interprets that as turning one direction, if B leads A it's going the opposite direction.

If the reverse switch on the ELS control works, but mechanically reversing the spindle doesn't, that means that the TI board can't tell which direction the encoder is turning.  Could be the encoder is bad and only outputting on one phase, it could be a wiring issue, or could be a bad input on the TI board.  If you have an oscilloscope you can check the signals at the connector on the TI board, if not then replacing the encoder is the next easiest thing to try after a visual exam of the cable and connectors between the TI board and the spindle encoder.

MMCCLENAHEN, 2024-01-31:
Thanks for all the help. The encoder seems to work fine as the display reads RPM in both Turn directions of the encoder.I have an Omnitron Encoder, hybrid stepper motor and a CL86T driver. When I change the driver Pulse Mode to CW/CCW(double pulse ) the stepper matches the movement  of the encoded. When the driver pulse   mode is set to PUL/DIR(single pulse) the stepper only turns Clockwise. Im guessing I should’ve had the driver in double pulse mode to start with.

dsgroninger, 2024-02-01:
Ahh, that goes back to my description of a quadrature encoder's signals as opposed to a directional encoder.  In a directional encoder, one signal is a square wave of pulses proportional to movement, while the other signal is hi for one direction of rotation  and low for the other, but steady while moving in one direction.  Sounds like the "double pulse mode" terminology matches up with quadrature type encoders and you have it all working.

Once everything is running correctly, I think you'll really like having this in place.  It's worked great for me for about two years now, never a burp once I got everything dialed in at the start.

Chucksbp, 2024-02-01:
Im glad you were able to get it working. Im sure you will enjoy yoyr ELS for years to come. I love mine!

closed 2024-03-10 by MMCCLENAHEN