clough42/electronic-leadscrew #283: Length Display Cable

Olazas, 2024-02-10:
Hi...  Any trick to get work well a long display cable??,,, 1,5 metres long for example

Thank a lot!!!!

Chucksbp, 2024-02-10:
My Display cable is close to 2 meters (6ft) in length and I have never had any issue with the display other than the first one didnt work at all. I used regular Earthnet twisted pair cables that used to come with electronic devices before wireless conectivity. I had a bunch of new cables so i actually used them for all my cabeling. i just cut the ends off and used one twisted pair for power positive and negative and a second pair for Data. the left overs i twisted around everything in case needed in the future. Mine hasbeen working fine for about three tears now.

Turner75, 2024-02-10:
Hi
I found that aethernet twisted pair cables work ok in a place where
electrical interference is low to moderate, just listen on an am radio for
an indication of RF ( radio frequency) noise, clicks and frying bacon!
Heavy interference places, eg, arc welding, high current switching,
requires screened cables to keep the signals to the display clean. A 1.5
metre cable is not much of a problem usually.

Olazas, 2024-02-12:
Thanks a lot!!!!