clough42/electronic-leadscrew #286: Panel doesn't light up

John5431, 2024-04-23:
So I had been sick all winter not playing with the lathe. So this weekend I went down and had a project to punch out. No threads or anything just a couple of collars for my  tractor. Turned the lathe on and bam the electronic leadscrew didn't light up except for a light behind the power button. Thought just push the power button nope nothing. Thinking the board is fried or am I missing something. 
John

tomvikse, 2024-04-23:
Check 5v power supply

John5431, 2024-04-23:
Nope opened the box and I could clearly see 5v was lit. Hmm opened the display box and vola the 4 pin connector is loose plugged it in all is good thanks
John

John5431, 2024-04-25:
Everything if fine now, just a loose connection on the display board.
John