clough42/electronic-leadscrew #288: ELS is great but want to make some code mods

SteveMaxPerry, 2024-06-17:
I don't have a real problem per say other than I'm not a good programmer. I see so many other uses for this system other than driving the lathe leadscrew, which is what my first ELS has been doing successfully for three years. 

But now I'm adapting one to a Chinese mill-drill to drive the mill's X-axis from its spindle. This allows for some great potential for simple power-driven feeding or driving a rotary table or even creating gears via hobbing. James has expressed some interest in hobbing previously.

But for now I want to A: Remove the lockout feature that prevents mode changes and B: Prevents reversing and C: Inhibits power on-off whenever the spindle rotation is being read. Being able to have on-the-fly control over the mill table at all times is different than doing threading or turning on the lathe.

Also, instead of the default feed rate of .005", I'd much prefer to start with the minimum feed of .001". 

These changes must be super easy to do for someone that knows code, that's not my specialty. 

Thanks!

SteveMaxPerry, 2024-06-18:
Thanks but I'm already committed to the hardware and future plans for it. I just need the changes as requested in my post. If you read my post again you'll see I do want to be able to program a fixed relationship between the spindle and stepper motor rotation. I'm already over halfway there.

Chucksbp, 2024-06-18:
Thats fine, I intended to do the same actuall I have a second complete ELS less th stepper motor. I didnt need to track with my spindle speed for any reason. so i went with this alternative motor and controller. Just thought I would offer my find. I wish i could help you with codes, but thats way beyond my paygrade! LOL! Im 75 years old and far from computer literate lett alone righting code. I do love the ELS on my Atlas lathe though, have had it for several years now. I had an inital issue shortly after i installed it. I had an encoder go bad and take out the first input of my launch board. but after finding a much better incoder all has been fine. 
   Good luck with your project!!

SteveMaxPerry, 2024-06-18:
My current ELS has transformed my 6" 1950's Craftsman lathe into a threading monster! I'm a few years younger than yourself and have owned this lathe since 1980. Made many upgrades to it such as an entirely new cross-feed dial (think Hardinge) and trued up the misaligned compound components. In parallel to the ELS is an Arduino-controlled rotary index for the spindle and to go along with that is a brushless motor milling/grinding attachment that bolts to the tool rest on the compound. This lathe will be with me "forever" if possible. 

I also have a nice cabinet style Atlas 12" lathe that's going to receive an ELS even though it has a good quick-change gearbox. I need ready access to metric threads so it's Bye-bye quick change. ELS is super-quick change! Thanks.

SteveMaxPerry, 2024-06-21:
@clough42 
Good morning James, I know you're busy but I don't know anyone else that's more qualified to help me other than yourself. My requests aren't to correct a problem but to customize the software a little for a unique application.  I wouldn't bother you if I could figure out the needed changes on my own. I've tried a few different changes and either the control panel buttons quit working all-together or nothing changes that I can detect. Issues #289 could use a little review too if you please. Adding the ELS hardware to my mill is opening up interesting options to manual machining. Thank you very much, SteveMaxPerry

clough42, 2024-07-25:
The reason the ELS firmware blocks mode and direction changes while the lathe is in motion is because doing so will ask the large, heavy mechanical system to change direction instantaneously.  Most steppers and servos will lose steps or panic if you attempt this, hence the protections built in to the firmware.

If you really want to try it, the test is on line 169 of UserInterface.cpp:

[image]

You could just change this to: if( true )

**I DO NOT RECOMMEND THIS CHANGE**

SteveMaxPerry, 2024-07-25:
I'm definitely willing to try it for my adaptation to the mill. As I'm only asking my stepper-driven axis to move at a controlled rate and not so much deal with motion accuracy a loss of steps isn't a problem. I've extensively driven my mill's table using that little pulse generator that's visible in one of my photos and have intentionally 'abused' the accel/decel motion to see how the system took the shock of no ramp up or down and I'm okay with the results. 

Can't wait to move forward with this project as there's lots left to do. Thank you so much.

SteveMaxPerry, 2024-07-26:
Thank you James, the little change works perfectly. 

Funny thing, when I started up the Code Composer I was looking directly at that line! I never would have guessed making that simple change would shut off the spindle rotation detect.

bogus105, 2024-08-24:
What about adding extra feature to the software - ENABLE switch. Making ENABLE inactive we can allow the software to accept inputs from UI to change various parameters. ENABLE feature to be accessed only if RPM == 0 (machine stopped).

SteveMaxPerry, 2024-08-24:
Isn't that the current firmware setup? Most UI requests are blocked unless there's zero RPM. 

Maybe give some examples of the type of UI inputs you'd want to make.