clough42/electronic-leadscrew #289: On-the-fly Encoder Selection, Using DRO Scale encoder output...

SteveMaxPerry, 2024-06-19:
Is there a documented way to select one or the other EQEP input? I'm considering using the output from one of my DRO glass scales as a signal source to drive my stepper on my mill's X-axis. 

And is there a way to configure the input to deal with the signals from the scale if they're slightly or significantly different from those that come from the typical 5V rotary encoder? 

The idea here is to be able to coordinate the stepper-driven X-axis of my mill in an angular motion relative to the Y-axis being driven by the hand crank. I can imagine that with a little fine tuning of steps I can generate close to 30, 45, 60deg angles rather easily. Other angles can be created through experimentation or careful calculations.

Thanks!

SteveMaxPerry, 2024-06-21:
A little update to my ELS mill project from a test earlier in the day: 

Success! The output of the DRO scale is perfectly compatible with the Launchpad board encoder input with no modifications. Adjusting the steps is absolutely necessary to gather a good match with the X-axis travel but that's not unexpected. 

I could really use some assistance with knowing how I can easily switch to one or the other encoder inputs via some control panel key presses vs uploading the firmware every time I need to swap encoder types. 

Let see if I can post a short video here:
Guess I can't. Photos won't tell the story but will have to do. 
[image][image][image]

clough42, 2024-07-25:
There isn't any provision in the code to change configurations on the fly.  The selection you speak of is a compile-time switch.  Modifying the firmware to be able to select a different encoder input (and different ratios) should be possible, but as with your other issue, you would need to be extremely careful about changing things while the system is in motion.  In most cases, that's definitely not something you want to do.

SteveMaxPerry, 2024-07-25:
I understand the ramifications and in the case of switching encoders it would be done when all machine motion has halted. 

The vast difference in the number of pulses per required motion of the rotary encoder vs the linear encoder means it's easier to 'calibrate' the two different styles of encoders if each one could be set up on its own channel. I can play with the numbers without having to compromise or favor one over the other. 

But how to do this switch/recompile? How do I find someone willing to help? It's way over my 'pay grade' lol. This mill will be so uniquely useable when I can sort out the control details. 

Thanks again 🙏🏻