clough42/electronic-leadscrew #291: Code Composer v12

kmanuele, 2024-08-08:
Has anyone gotten this to compile with current version of CCS (12)?

Or, the latest version that will compile?

thanks

clough42, 2024-08-08:
The code was developed in v9, which you can download from the TI web site.

I know there are people who have succeeded with newer versions, but they
changed the internal build infrastructure.

James

kmanuele, 2024-08-09:
ok. V9.3 compiles just fine. Thanks for prompt response.

Got spooked when v12 was looking for a MAKEFILE.

K

bogus105, 2024-08-25:
Guys, same issue here. Somehow i can't find v9 of CCS on their site. Pls, can you post the link here?

tomvikse, 2024-08-25:
https://www.ti.com/tool/download/CCSTUDIO/9.3.0.00012

bogus105, 2024-08-25:
Wow! This is FAST response. Thank you @tomvikse.

In the meantime i have found the following workaround:
- CCS i'm using is version: 12.8.0.00012
- compiler version was 'too new' to compile original @clough42 software

1. In CCS go to: Help -> Install New Software...
2. In "Work with: " drop down menu pick "Code Generation Tools Updates - https://software-dl.ti.com/dsps/dsps_public_sw/sdo_ccstudio/codegen/Updates/p2win32"
3. Then deselect (un-tick) "Show only the last versions of available software" box
4. Then open drop down menu under "TI Compiler Updates" - you will see long list of available packages
5. Pick "C2800 Compiler Tools", version "18.12.1`" and install it.

After installation and rebooting i have opened the project and run build with no errors and no warnings!
For the moment i can't tell more as i don't have a board at hand but it lloks promising as all compiled with no errors:)

Thank you for suuport /bogi
7.