clough42/electronic-leadscrew #293: Breadboard test with servo...

Andymac4, 2024-10-21:
Does anyone know the PIN allocations on the Ti board that the Phoenix connector links to...
I have a servo to test on the bench and haven't been able to get the PCB daughter card from James yet. I can see where the "STEP-" to "ALM+" connections go but the other two connections have me confused. I can't work out where the "+" and the "ALM-" eventually arrive on the Ti board. Hope I'm not talking rubbish so please help if you can.

Thanks,
Andy

conach648, 2024-10-21:
I assume that the PCB daughter board you refer to is the board James calls the ELS Boost board.
You can not connect the control panel to the TI board without the Boost board. The boost board is required to convert the 5v data from the control panel to 3.3v required by the TI board. If you connect the control panel to the TI board without the Boost board the TI board will be destroyed. Ebay has the Boost board in stock.

Andymac4, 2024-10-21:
Hi,
Thanks for your reply but I think you missed the point of the question. Like James did at the start of his development journey on a breadboard I am using a bidirectional boost to convert 3.3v signals to 5v and then back down to 3.3v on the way back to the Ti board. My question still stands.

I have thought some more and maybe they are 5v and GND. In that case I won’t find a GPIO pin is the answer. I need to deliver 5v and GND to the servo. Do you agree?

Thanks,
Andy

conach648, 2024-10-21:
It was not clear that you were using a bidirectional boost. To answer your question: I do not know what type of stepper you are using but either way you need to supply +5v and a ground to the proper pins on the driver. You should not need to connect anything else but the step and direction and enable to test the motor. I see no need to connect the alarm to bench test. The motor should not go into an alarm if there is no load on it unless you run it too fast and over speed it or you have a load connected and apply an external force to the motor to attempt to stop it while it is receiving step and directional information. That could occur on the lathe if you run the carriage into a stop without a limit switch to disconnect the step signal. Either way if you over speed or intentionally stop the motor by external force you need to immediately switch off all power to the stepper. So make sure you have a switch on the power line from the stepper power supply.

paul15985656, 2024-11-07:
I know it's a bit late and I can't help directly but have you looked at the projects on circuit maker? You should be able to work out wiring information from the projects there.

The GPIO pins used will also be somewhere in the CCS project too, this is how I worked it out back in August 2019 to build it up on a breadboard, although without an alarm input.