clough42/electronic-leadscrew #295: Cannot permanently flash Ti Launchpad

Gouge61, 2024-10-31:
I am on my second Launchpad.  The first did not flash, or did not indicate so, when removing 5v power and then reapplying.  Got only small LED at left of control panel.  I also bought a new ELS Boost card, configured and attached to the new Launchpad.  Nothing.  I am using CCE v9 on a Windows computer.  I can run the program in CCE, either debug or release mode, witness "Clough42 ELS" and v1.4 on my control panel, followed by "0.005" and the ability to change numbers and functions, including the Off button.  When I pull out the 5v power and put back in, nothing but the small LED.  CCE is telling me that the 110 debug is running, same as James.  Whatever flashes to the card does not remain, even though I am using "release" in CCE.  I am not a software engineer.  Thoughts?  Similar issues?  All continuity checks of the Wiki have been tried and all show continuity, including the cable.

Chucksbp, 2024-11-01:
When I programed my boards, i did them on my PC but didnt use the 5 volt power supply. I reinstalled the three jumper plugs on the main board an d just used the supplied wiring connector supplied with the TI Launch board.  Then removed the three jumpers and used the 5 volt sspply for actual use. 
 One thing I would check and double check is the positions of the switches you had to move on the TI Launch board. My switches were very difficult to move, I had to flip them back and forth multiple times to get the sticky out of them left over from the cellophane covering. 
  If all else fails you may want to try and remove code composer completely and reinstall it. 
are you using the actual flash button? looks like a page with an arrow pointing down. 
  Im no expert either far from it, just throwing out some ideas. Its rather doubtfull yo have two bad pieces of Hardware doing the exact same thing. I personally had no issue programing the board luckily. I did watch the videos over and over again. You do have the right 9.0 version of code composer, thats what worked for me.

Gouge61, 2024-11-01:
Thank you for your comments!

I have tried everything you listed, and in addition, tried to load the code with my Mac running v10.  I get exactly the same results.  I agree now that it is doubtful both my Launchpads are bad.  My booster card is new and passes all continuity checks.  I am not a software or electrical engineer, so I, in my feeble mind, and slogging through this dimly…

If you know anything new, please pass it along!

Chucksbp, 2024-11-03:
In code composer, in the very lower right hand box, under descriptions, does it show any problems??? if so those have to be fixed.

Chucksbp, 2024-11-03:
You said you are using code composer 9 ??? the one i used is 9.3.0 specifically. 
If you right click on the icon it will show the version. I had issues finding a version that worked on my computer. unfortanatly it was three years ago and I really dont remember what the issues were. Like you, im a hands on guy, and the computer part isnt my strong point! I really wish someone with some knoledge would get on here and help you!! I feel your frustration. when you get it going please let me know what the problem was. I have two Launch boards programe just in case of failure. I lost one of my encoder inputs do to the encoder failure so i got a second board for backup. at 75 years old two shoul do what time i have left, although I do have a third one unprgramed along with a spare Clough 42 interface and spare encoder.

Sprint77, 2024-11-16:
> Were you ever able to get your launch board programed??

I just got my board yesterday.  So I am still gathering  parts

Gouge61, 2024-11-16:
No, I was not. The CCE program indicates the flash takes place, but if I remove 5v power from the card, the controller does not come on again with “Clough42” etc. very frustrating. I have purchased two cards and booster cards, both with the same problem. Sent from my iPhone