clough42/electronic-leadscrew #296: Nema 34 Stepper No movement.

Helkar2k, 2025-01-19:
I'm using the control panel and boost kit directly from Clough42 along with the TI Launchpad.

Software version 1.4.0 

--------Here are the other components I am using----

Stepper: Model is 86HSE156; Holding Torque:12N.m 1700oz-in; Rated Current:6A; Peak Current:8A; Phase:2-Phase; Size:86x86x156mm; Step Angle:1.8 degree; Motor Lead Wire: 4-Wires; Encoder lines:1000; Shaft diameter: 14mm

Stepper Driver: Model is 2HSS86; Type:2-Phase Hybrid Stepper Servo Driver; Frequency:0-200KHz; Insulation resistance:>=500MΩ; Voltage: AC 24-70V or DC 30-100V input

DC Stepper Power Supply:  S-400W-70V-5.7A

DC Board Power Supply: DC 5V Din Rail Slim Switching Power Supply 15W 2.4A

Rotary Encoder: Model: E6B2-CWZ6C Input voltage: DC5 ~ 26V(open-collector output type) Output phase: A, B, Z phase

----Project details------

All systems appear to be working currently , I am getting consistent and stable RPM readings from the Rotary Encoder in both CW and CCW rotations.
I AM using all provided cabling with the stepper motor.

LINK TO VIDEO SHOWING ISSUE: https://youtube.com/shorts/p1xK6Pr9r_s?feature=share

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

----ISSUE----

 I am getting no movement from my stepper motor. When unpowered it does rotate freely and when energized it has braking(holding power) on.

---Diagnostics I have performed so far---

All continuity checks on the boost board as per James instructions.

Continuity tested from both pin plugs (via screws terminals and the end of plug) on the cable used for connecting boost board to Stepper Driver.

Changed microsteps value on driver via DIP Switches , no affect.

Confirmed 5v DC power on (DIR +) (EN +) and (PUL +) pins on driver.

Confirmed 70v DC power to stepper Driver pins. 

Confirmed all wiring pinouts to be correct.

Motor power Leads hooked directly to Stepper Driver with no Additional extension.

Verfied S4 Switch moved to the LEFT postion(towards Center of board).

Verfied BOTH S3 Switches ro the RIGHT position(towards Edge of the board).

Verfied ALL 3  Power Jumpers removed from JP1 JP2 AND JP3.

Confirmed TI Launchpad power supply is true 5v DC and has no grounding except its power supply.

120v Power is grounded ONLY to Ground INPUT on 70v Power Supply.

Enclosure is NOT Grounded as it is ABS Plastic.

Shielding from Rotary Encoder Cable is NOT Grounded.

Shielding from Stepper Motor Cable is NOT Grounded.

CincyFlyer, 2025-01-19:
I had similar issues, and because I'm lazy, I didn't thoroughly read your post.
But my issue was simply that the stepper controller wanted to see a 25 volt signal by default! It's a newer version of one that I've used before, and it seems some new control interfaces are that crazy high voltage (relatively speaking). 
There is a tiny switch to change that to five volts, which the ELS board can output. 
**_It was not documented in the printed instructions, but was in the latest online edition._**
See if you have that issue, it's a quick fix if you do.

Helkar2k, 2025-01-19:
> I had similar issues, and because I'm lazy, I didn't thoroughly read your post. But my issue was simply that the stepper controller wanted to see a 25 volt signal by default! It's a newer version of [...]

I just looked in the documentation I can find for this controller and it does say it supports between 5v-24v for inputs. I pulled the controller out of the enclosure and checked every angle on it and do not see a switch for changing that setting anywhere, so I'm guessing its not that.

Helkar2k, 2025-01-20:
[image]
[image]
[image]

Pulled apart the Stepper Driver to verify there was no selector switch for input voltage.

---Setup LEDs to test output of Stepper Driver Control Pins. Here's my findings---- 

Note: LEDs are powered via +5vDC Pin from the ELS Boost Board and Hooked to each output via 220 OHM resistor.

Green LED Hooked to ENA- Pin activates when Control Panel "Power" Button is switched "OFF"

Blue LED Hooked to DIR- Pin Actives when Rotary Encoder is moved CW and will stay lit until Encoder is moved CCW. LED will NOT light up when moving in CCW at all.

Red LED Hooked to PUL- NO OUTPUT at all. - Tried Inverting this pin via software change but made no difference.

Leaning towards bad GPIO Pin on the Launchpad?

CincyFlyer, 2025-01-20:
Have you any method of testing the driver and thus the stepper?
If you can test that separately (I have a little pulse generator) then you
can concentrate on the board.
James' daughter board could be bad, or the pin on the TI that feeds it. You
could technically switch to the other pin header to eliminate the TI
Launchpad board from consideration; don't do so unless you are fairly
conversant with such things. The boards aren't super expensive, but are
sometimes hard to find, and may take a while to reorder.
I have several of these on hand (well, several Clough42 kits, TI boards,
power supplies, and hybrid steppers), so I could just swap out if I have an
issue.
Certainly start from scratch and go from the encoder to the board to the
output to make certain all is well.
When I had an issue I broke and reconnected every single junction.

Helkar2k, 2025-01-20:
> Have you any method of testing the driver and thus the stepper?
> If you can test that separately (I have a little pulse generator) then you
> can concentrate on the board.
> James' daughter board could [...]

I am fairly new to the whole microcontroller deal but I do have an Arduino Uno I could probably set up to drive the stepper. I did look at those Stepper Motor controllers on Amazon , wasn't sure if they pretty much work for any driver or not(Is that what you have?). I did order a cheapo Oscilloscope to to some more testing too.

clough42, 2025-01-23:
Wow.  Thank you for all the detail.  That's very helpful.

Some thoughts, based on what you wrote:

- You said the RPM display is correct and stable.  This means the encoder is working, connected, and configured properly.
- You said the rotor of the stepper motor locks when you switch on the ELS and unlocks when you use the power switch to turn it off.  If I understood correctly, this mean the Enable line is connected properly, the driver is responding to the 5V signal, the stepper power is good, and you have it wired correctly (at least one phase, but probably both).  I just want to confirm: When the ELS is ON (displaying a feed or pitch) the motor is locked, and then the ELS is OFF (feed/pitch area blank, displaying only RPM) the motor is unlocked?  If so, the enable is working.  I initially suspected it might be miswired, or it might be inverted in the driver settings, but it sounds like it's working.
- The tests you did with the LEDs are good.  It sounds like you (re)confirmed that Enable is working, and you can see the Direction pin behaving as expected, but you're not seeing any pulses on the Step pin.  The pulses on the step pin are only 5 microseconds, so you likely would not be able to see them, even if they're there and working properly.

If I understood everything correctly, I think the only issue you have is that the stepper driver is not responding to the step pulses.  I can think of a few hypotheses worth checking:

1. You might have a bad pin on the LaunchPad (as you suspect), a bad transistor on that channel of the boost board, or a bad solder connection somewhere (your wiring, or one of the PCBs).
2. The driver might be unable to respond to the pulses for some reason.  Most drivers have a minimum pulse width specification.  Usually, it's around 1us, so 5us should be plenty, but it would be worth checking to see what they specify.  What you posted says 200khz, which would be a 2.5us pulse width for a square wave, but it would be good to be sure.
3. There might be something wrong with the driver.

If you have an oscilloscope, probing the step line to see if pulses are arriving would be a good way to find out what's happening for sure.

Another thing you could try would be to invert the step pin and see if your LED lights.  Just uncomment line 76 in Configuration.h:

#define INVERT_STEP_PIN true

If you uncomment that and re-flash the board, then the Step pin should always be active and should pulse off for 5us to take a step.  You don't want to run it that way, but it would confirm if the LaunchPad pin and the transistor on the boost board are operating properly.

With it in this state (always asserted) you could take the step wire out of the connector on the driver and tap it on the connector to take steps.  That would confirm if the voltage produced on that pin by the ELS is sufficient to command the driver and would let you know if the driver has an issue.

Helkar2k, 2025-01-24:
> Wow. Thank you for all the detail. That's very helpful.
>
> Some thoughts, based on what you wrote:
>
> * You said the RPM display is correct and stable.  This means the encoder is working, connected, and [...]

I did invert the STEP pin via the programming earlier and didn't seem to get anything via that either. I do have a new Launchpad coming on Monday so hopefully that fixes everything up or at least gives me a sanity check with what I've tested so far. 

I have jumped the Driver to ground via the STEP Pina and do get movement, so I'm fairly confident in the driver being ok.

I did get a small single channel scope but this is my first time using one so I have to research a bit more to make sure I'm using it correctly.

clough42, 2025-01-24:
Sounds great.  The key question that I think remains is whether the
LaunchPad pin is bad, or if the transistor on the boost board is the
issue.  If you have a new LaunchPad board coming, that should settle that
pretty quickly.

James

Helkar2k, 2025-01-25:
> Sounds great.  The key question that I think remains is whether the
> LaunchPad pin is bad, or if the transistor on the boost board is the
> issue.  If you have a new LaunchPad board coming, that should settle that
> pretty quickly.
> 
> James
> […](#)

Well just flashed the new Launchpad, set all the switches to their proper positions and removed all 3 jumper pins from the debugging  area and installed it. No such luck , still no response from the stepper. Is there a straight forward way to test the transistor on the boost board? Has to be something stupid and simple at this point...

Helkar2k, 2025-02-04:
ISSUE SOLVED!! I bought a new Boost Board, soldered everything on , plugged it in and everything works flawlessly!

closed 2025-02-04 by Helkar2k