clough42/electronic-leadscrew #297: No display shown

oracleglassarts, 2025-03-04:
Hi all.  I just got a new kit soldered together.  After programming the launchpad, everything seemed to succeed, but to my surprise the display didn't show CLOUGH42 after pressing the resume button in CCS9.

The display came with a right angle connector already, but I did reflow all of the solders to try to troubleshoot.  I also tried a different cable and verified the connections.

I need to make a cable for the stepper, so the rest is untested, but I'm focusing on the screen for now.

Is this a bad display board like others have had?  Is this on me?  I got the kit through James on ebay.  I didn't have any issues putting it together or do anything wrong, it just doesn't work as far as I can tell.

[image]

oracleglassarts, 2025-03-08:
Well, I got two new TM1638 modules and they didn't do anything differently.

Chucksbp, 2025-03-08:
Did you get all of the switches properly moved on the TI Launch board?? they can be quite finiky to move. sometimes the glue from the tape stops their complete movement.  aldo did you removethe jumpers??? 
If yes to both, check continuity of the cabeling from board to board. I am also wondering if you actually programed the launch board. what i mean is did the programing go through?? 
slight chance  the TI board could be bad????

oracleglassarts, 2025-03-08:
I did get the switches and toggled them many many many times to rule out sticking or not contacting all the way.

Jumpers are removed.

The continuity checks are complete from end to end with cables connected.

5vdc at the tm1638 key & led board, tried multiple tm1638s as well as the original clough42 display.

The code appears to load correctly and I am able to read data back in the memory browser of code composer.

I have been in contact with James on ebay but it's just been a really slow process.  It takes a day or two for him to respond each time which is fine, but it's one test at a time and I've ran through them all already.

Chucksbp, 2025-03-08:
Sounds like you have done most everything, it appears that you have power and ground to the display. almost sounds as though the Launch board may have issues. I had an issue shortly after finishing ny project, it would no longer sense RPM. I dont what caused what but i replaced the encoder and then used the second input on the launch board. i also noucht a complete new kit and launch board for a backup. another fellow had the same issue at the exact same time.. I will say it may be a struggle now but the finished product is so woth it in my opinion. im sure you you will get it going, and believe me you will love it.

oracleglassarts, 2025-03-08:
This has just been a nightmare.  I started one about a year ago and had the encoder issues.  I got the encoder working by buying a different encoder, but never installed it on my lathe.

Everything from that got lost in a box and now I've started over with new components.  At this point my opinion of the project has soured and I just want something that works after all the wasted time and money spent trying on this two separate times.

I have a service engineering background and have worked in automation for the last 3 years, this shouldn't be hard for me, but every aspect of the project has been riddled with problems on both iterations.

Chucksbp, 2025-03-10:
I hope you solve the problem soon. let us know what you find to be the issue. My build wemt fairly well. my first display didnt work, it was the run of the mill displays. I bought a few from Amazon and got a good one on the second try. everyting worked well for a couple weeks then the encoder failure that took ouy the first input of the Launch board. I got a much better encoder from Amazon and used the second input and all has ben well. Im 75 years old have some electronics background but mostly automotive, I have an associates degree in electronics but thtas from 1969 we hstill had vaccuum tubes! LOL!!

oracleglassarts, 2025-03-10:
Thanks, I appreciate it.  It's more or less on hold now.

I'm out of ideas, so I'm at the mercy of someone smarter than I to suggest things at this point.

I'm strongly considering just cutting my losses by returning whatever I can and buying a quick change gearbox.