clough42/electronic-leadscrew #300: Grizzly G0602 Lathe ELS Issues

ODmechanic92, 2025-04-29:
I completed installation and am currently validating the system. General operation is in order however, I have encountered the following two issues and was hoping to get some feedback on them if anyone has time. Thank you in advance.

1) During testing/validation i discovered a discrepancy between the selected feed rate and the actual feed rate.
Ex: ELS display set to .010"/rev feed rate, measured feed of .006"/rev plus or minus .001"
Using the online material both here on github and Clough's youtube playlist i verified my gear ratio at 4:1 (36T servo pulley to 72T gearbox input pulley, G0602 gearbox set to C-1 for an additional 2:1 creating a 4:1 ratio) and confirmed that the ELS program is set to 4 micro steps with a resolution of 1000 microsteps. Not sure of where to go from here.

2) Not sure if this is something you have on your lathe but it seems my gearbox has about 1/3 of turn of play in the output when changing directions. Is this normal on these lathes or is this potential wear/mfg defect?

tomvikse, 2025-04-29:
1. your stepper driver is set to xxx steps?
2. backlash is normal

ODmechanic92, 2025-04-29:
1) the stepper driver is set to the default 1000 pulse/rev.
2) I understand that a some amount of backlash in any gearbox is to be expected but 1/3 of a turn of the input before any output? Seems like a lot of backlash

tomvikse, 2025-04-29:
Leedscrew tpi is correct set up?

Got 1/4 Backlash and is not a problem..

ODmechanic92, 2025-04-29:
Yeah. I'm on the same platform as clough when he developed the ELS. I did confirm the leadscrew is 12 TPI

ODmechanic92, 2025-05-02:
Just a couple thoughts and trouble shooting steps to help narrow down the issue.
1) Verified all pully tooth counts (all correct)
2) Re checked servo parameters.
-counts/rev set to 1000
3) Adjusted ELS Software Microsteps and Resolution
- microsteps set to 4 (per explanation in software)
- resolution set to 1000 (same reason)
4) Verified gear reduction in lathe gearbox
-setting C-1 is verified 2:1
5) Isolated any electrical noise
-installed ferrite chokes on all signaling cables and some wires

I'm still not getting accurate thread counts or advance/rev. Any suggestions?